Livox-SDK/livox_ros2_driver #8: Using the ros clock as timestamp

tigelbri, 2021-04-16:
Hi,

is it possible to use the ros clock instead of the package timestamp for publishing or wouldnt it make sense to have the possibility for that?

Great Thanks is advance

Livox-SDK, 2021-06-17:
Hi tigelbri,

It is possible, but we did not develop based on Ros clock
Reason: The ros time is the moment system receiving point cloud, it is easily to be effected by data transferring way. This will make the ros bag not precise and not even;(if lagging, at such a moment you will received a big amount of point cloud)
I guess what you need is Lidar synchronizing with multiple sensors. I would suggest you synchronized the system time with ptp(1588)/gps time first.

syt011019, 2025-04-10:
Are you solve this problem? how can use ros time?

TokyoWarfare, 2026-02-06:
> Hello, my camera uses the ROS system time as the timestamp for publishing. How can I make the timestamp of my images and the timestamp of the LiDAR be in the same time domain and synchronized? Can anyone offer some assistance? Thank you very much.

I would say the best way would be to use PTP or if your cammera supports PPS sync.

In my specific case what I do is to have a GPS module that outputs PPS + RMC time. The pps pulse goes directly to the livox converter 2.0 And the RMC (via NMEA message) data goes trough serial to the PC where it is used by the livox driver and the lidar is timestamped with RMC time.

For the camera, since I'm using a webcam I'm simply using the ros topic /navsatfix, extract the timestamp from the header and stamp my frames with it. I've just added the processing dealy to the stamp.

I'VE NOT YET TESTED it, I'm finishing the rig. As said, since this is webcam has no seanc ports nor PTP support.  I would say most if not all industrial cams suport PTP or PPS sync which is the pro way to handle this.

I'll soon update the latest fix to the camera driver that supports the timestamping fix plus compression.

Check here in 24-48h
https://github.com/TokyoWarfare/stereo_uvc_camera_mono_support/tree/gnss_timestamp
the original driver is not mine.