Livox-SDK/livox_ros2_driver #10: Imu burst mode sending

tigelbri, 2021-07-05:
Hi,
I have a question about the IMU on the livox horizon. When I do a plot of the IMU data then Iam reveicing the data in a kind of burst mode where I get 20 dates in about 2 ms and then 100 ms pause and repeat. Is this the way how it should work or do I missed something on my side?

It seems like that the imu data is received every time a scan is received. So is there a way to get the imu data more frequently?

Great Thanks in advance

Tristan

Livox-SDK, 2021-07-12:
Hi, tegelbri,
    You can create a new thread, and poll the imu data in your needed frequency.

Livox-SDK, 2021-07-12:
[image]

tigelbri, 2021-07-16:
I tried it.
The IMU now gets its 200 Hz, but Iam receiving some lags in the data.
[image]

TokyoWarfare, 2025-11-19:
> I tried it. The IMU now gets its 200 Hz, but Iam receiving some lags in the data. [image]

Did you manage to solve this? I'm facing exact same issue. Stuck ad 50hz rate.