Livox-SDK/livox_ros2_driver #11: Does it work for ros2 foxy?

Sollimann, 2021-11-16:
Does the driver work with ros2 foxy?

We're thinking about buying Livox mid-70 for our robots, but need to know if it is compatible with ros2 foxy.

SeungRyeol, 2021-12-03:
I am also curious about the working in ros2 foxy.

jarekkt, 2021-12-08:
Yes, it does work in ros2 foxy

livox_ros2_driver_node-1] [INFO] [1638949297.767650580] [livox_lidar_publisher]: Livox Ros Driver Version: 0.0.1
[livox_ros2_driver_node-1] [INFO] [1638949297.768077732] [livox_lidar_publisher]: Data Source is raw lidar.
[livox_ros2_driver_node-1] [INFO] [1638949297.768102298] [livox_lidar_publisher]: Config file : /home/jarek/Projects/vcs/self-driving-system/integration_3/src/launch/../config/livox_lidar_config.json
[livox_ros2_driver_node-1] [INFO] [1638949297.769315113] [livox_lidar_publisher]: Init lds lidar success!
[livox_ros2_driver_node-1] [INFO] [1638949299.318707242] [livox_lidar_publisher]: livox/lidar_0TFDFCE00503741 publish use PointCloud2 format

YHanG505, 2024-09-09:
What about humble? What should I do to make it work for ros2?