Livox-SDK/livox_ros2_driver #13: can not be used when making a shared object

RENyunfan, 2021-12-06:
在ROS2 Foxy中编译时报错如下

```bash
/usr/bin/ld: /usr/local/lib/liblivox_sdk_static.a(device_discovery.cpp.o): relocation R_X86_64_TPOFF32 against symbol `_ZGVZN6spdlog7details2os9thread_idEvE3tid' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/liblivox_sdk_static.a(livox_sdk.cpp.o): relocation R_X86_64_PC32 against symbol `stderr@@GLIBC_2.2.5' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/livox_ros2_driver.dir/build.make:462: liblivox_ros2_driver.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:80: CMakeFiles/livox_ros2_driver.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< livox_ros2_driver [8.29s, exited with code 2]

Summary: 3 packages finished [11.9s]
  1 package failed: livox_ros2_driver
  1 package had stderr output: livox_ros2_driver
```
使用的Livox-SDK和livox-ros2-driver均为当前最新版本。
请问有什么好的解决方案吗。

jarekkt, 2021-12-08:
This is a problem caused by different compilation options in SDK and in driver.
You would need to make this modification in Livox-SDK/sdk_core ( has to be built same way as driver - as position independent code - '-fPIC' needed )

From:
target_compile_options(${SDK_LIBRARY}
        PRIVATE $<$<CXX_COMPILER_ID:GNU>:-Wall -Werror -Wno-c++11-long-long>
        PRIVATE $<$<CXX_COMPILER_ID:AppleClang>:-Wno-unknown-pragmas -Wall -Werror -Wno-c++11-long-long>
        PRIVATE $<$<CXX_COMPILER_ID:Clang>:-Wno-unknown-pragmas -Wall -Werror -Wno-c++11-long-long>)     

to:

target_compile_options(${SDK_LIBRARY}
        PRIVATE $<$<CXX_COMPILER_ID:GNU>:-Wno-missing-field-initializers -fPIC>
        PRIVATE $<$<CXX_COMPILER_ID:GNU>:-Wall -Werror -Wno-c++11-long-long>
        PRIVATE $<$<CXX_COMPILER_ID:AppleClang>:-Wno-unknown-pragmas -Wall -Werror -Wno-c++11-long-long>
        PRIVATE $<$<CXX_COMPILER_ID:Clang>:-Wno-unknown-pragmas -Wall -Werror -Wno-c++11-long-long>)

AsiDiaz, 2022-01-14:
Hi @jarekkt,

I made the modification you mentioned, but I still have the same problem.
I am building it in Ubuntu 20.04 with Foxy.

jarekkt, 2022-01-14:
Make sure you actually compile what you think you compile :-)
The code - when taken directly - pulls from github fresh copy of the driver
You need to force it to take yours - see part of my build script.
It forces use of previously built (and fixed) SDK (install directory) - this prevents local fetch of SDK

build_livox() {
    if [ -f /opt/ros/foxy/setup.bash ]; then
       source /opt/ros/foxy/setup.bash
    else
       source /opt/ros/foxy/install/setup.bash
    fi

    export COLCON_EXTENSION_BLOCKLIST=colcon_core.event_handler.desktop_notification
    export livox_sdk_ROOT_DIR=$ROOT/install/livox_sdk

    cd $ROOT
    colcon build  --packages-select  pcl_msgs
    colcon build  --packages-select  pcl_conversions
    colcon build  --packages-select  pcl_ros
    colcon build  --packages-select  perception_pcl
    colcon build  --packages-select  livox_sdk
    colcon build  --packages-select  livox_sdk_vendor livox_interfaces livox_ros2_driver
}

babatur, 2022-03-31:
> This is a problem caused by different compilation options in SDK and in driver. You would need to make this modification in Livox-SDK/sdk_core ( has to be built same way as driver - as position [...]

babatur, 2022-03-31:
That's great，I have solved the problem using your method

FPSychotic, 2022-04-15:
I had this same problem, I cannot understand any of the two responses to fix it. I'm not expert, so no idea where use or how that script. In the first response neither say if after modify I need recompile or what. So honestly is not very helpful for normal users with low linux knowledge. Some customers will now linux and others not, please make clear the steps.

I think both ways to fix, if do, will make issues if you want run in ROS and in ROS2 too, I think you will only can run or one or the other, but not in both.

Checking the cmakelists, fortunate of me god is with me and  in "<yourworksapace>_ws/src/livox_sdk_vendor/cmakelist.txt" change this line
option(BUILTIN_livox_sdk "Force built-in version of livox-sdk" OFF)
to
 option(BUILTIN_livox_sdk "Force built-in version of livox-sdk" ON)

It looks that build the right SDK version into your workspace so as further benefict you will can use the system one or ROS melodic o noetic.

it allowed me compile.