Livox-SDK/livox_ros2_driver #17: 3 issues running or using the node.

FPSychotic, 2022-04-15:
LIVOX AVIA, EHTERNET, NO SYNCHRONIZED, ROS2 FOXY ,UBUNTU 20.04, X64,GPU NVIDIA

Hi, after fix my building installation as said here https://github.com/Livox-SDK/livox_ros2_driver/issues/13#issuecomment-1100194401
the node works as expected I think, but I have the problem no TF is published, livox_frame is there because it is in RVIZ cfg. but it is not being published, that is problematic , and make me dude if other things are working , as the double return, IMU topic, or any other feature.

I noted there is some features from AVIA that I cannot see how make them work, they look no supported, triple return, repetitive mode and seeing the IMU topic in rviz2, there is not IMU TF, and not sure if the topic is even publishing.
AVIA has more of a year in the market is weird in ROS looks a mid-40 in features that can be used. That is not good, I bought this lidar to be used only in ROS/ROS2 and obvioulsy I want to use all its features. Please add them.

I'm surprised , as I needed those missing functions, will they be added?

The third issue is when I connect the AVIA via ethernet, I lose my internet from WIFI, how I can keep my WIFI internet working while using the livox? In windows I can use both

Thanks.

Funktiona, 2022-05-08:
@FPSychotic have you had any success with the TF problem, currently have the same

FPSychotic, 2022-05-08:
> @FPSychotic have you had any success with the TF problem, currently have the same

No mate I didn't, but no even a answer...

Funktiona, 2022-05-09:
@Livox-SDK Is this something you have a solution to? :)

Y-Roman, 2024-03-04:
I have a mid360 Lidar connected to Ubuntu 18 on Jetson Xavier. I'm able to connect using ethernet but I can't access Internet using WIFI. Any solutions to use both simultaneously?