Livox-SDK/livox_ros2_driver #18: ROS2 Humble

SethFarrell, 2022-06-14:
Using Docker based on an Ubuntu 22.04 Image and setup with ROS2 Humble, everything seems to work fine in a container using these (no livox).

Are there any plans to update Livox for this new distribution?

I get this error when building the docker file (after cloning the livox repo and building with colcon):

"""
Cloning into 'livox_ros2_driver'...
Starting >>> livox_interfaces
Starting >>> livox_sdk_vendor
Finished <<< livox_interfaces [15.1s]
--- stderr: livox_sdk_vendor
Cloning into 'livox-sdk-v2.3.0'...
HEAD is now at 7cf759a remove dependency on APR library
In file included from /opt/sensor_ws/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp:25:
/opt/sensor_ws/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.h:46:8: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
   46 |   std::shared_ptr<std::thread> thread_;
      |        ^~~~~~~~~~
/opt/sensor_ws/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.h:30:1: note: ‘std::shared_ptr’ is defined in header ‘<memory>’; did you forget to ‘#include <memory>’?
   29 | #include "noncopyable.h"
  +++ |+#include <memory>
   30 | 
/opt/sensor_ws/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp: In member function ‘virtual bool livox::ThreadBase::Start()’:
/opt/sensor_ws/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp:34:3: error: ‘thread_’ was not declared in this scope; did you mean ‘pthread_t’?
   34 |   thread_ = std::make_shared<std::thread>(&ThreadBase::ThreadFunc, this);
      |   ^~~~~~~
      |   pthread_t
/opt/sensor_ws/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp:34:18: error: ‘make_shared’ is not a member of ‘std’
   34 |   thread_ = std::make_shared<std::thread>(&ThreadBase::ThreadFunc, this);
      |                  ^~~~~~~~~~~
/opt/sensor_ws/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp:26:1: note: ‘std::make_shared’ is defined in header ‘<memory>’; did you forget to ‘#include <memory>’?
   25 | #include "thread_base.h"
  +++ |+#include <memory>
   26 | #include <thread>
/opt/sensor_ws/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp:34:41: error: expected primary-expression before ‘>’ token
   34 |   thread_ = std::make_shared<std::thread>(&ThreadBase::ThreadFunc, this);
      |                                         ^
/opt/sensor_ws/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp:34:68: error: left operand of comma operator has no effect [-Werror=unused-value]
   34 |   thread_ = std::make_shared<std::thread>(&ThreadBase::ThreadFunc, this);
      |                                                                    ^~~~
/opt/sensor_ws/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp: In member function ‘virtual void livox::ThreadBase::Join()’:
/opt/sensor_ws/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp:41:5: error: ‘thread_’ was not declared in this scope; did you mean ‘pthread_t’?
   41 |     thread_->join();
      |     ^~~~~~~
      |     pthread_t
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-c++11-long-long’ may have been intended to silence earlier diagnostics
cc1plus: all warnings being treated as errors
gmake[5]: *** [sdk_core/CMakeFiles/livox_sdk_static.dir/build.make:104: sdk_core/CMakeFiles/livox_sdk_static.dir/src/base/thread_base.cpp.o] Error 1
gmake[5]: *** Waiting for unfinished jobs....
gmake[4]: *** [CMakeFiles/Makefile2:242: sdk_core/CMakeFiles/livox_sdk_static.dir/all] Error 2
gmake[3]: *** [Makefile:136: all] Error 2
gmake[2]: *** [CMakeFiles/livox-sdk-v2.3.0.dir/build.make:86: livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0-stamp/livox-sdk-v2.3.0-build] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/livox-sdk-v2.3.0.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< livox_sdk_vendor [18.4s, exited with code 2]
"""

Marcus-D-Forte, 2022-09-30:
Pehaps you could take a look into:

https://github.com/Livox-SDK/livox_ros2_driver/pull/20

SethFarrell, 2022-10-01:
Yep I found that issue awhile back. I ended using their changes in my own forks for a personal project. Thanks

closed 2022-10-01 by SethFarrell