Livox-SDK/livox_ros2_driver #19: ros humble

hongsheng2000, 2022-06-27:
我在ros humble环境下编译，发生了错误
`livox_ws/src/livox_ros2_driver/livox_ros2_driver/lddc.cpp:323:23: error: no matching function for call to ‘rclcpp::Publisher<pcl::PointCloud<pcl::PointXYZI> >::publish(PointCloud&)’
  323 |     publisher->publish(cloud);
`

liyifan2002, 2022-08-01:
#20 this may help

TZECHIN6, 2022-09-22:
### Same error here, below is the environment setup and error reproduce procedures:

Environment:
- Jetson Xavier AGX Developer Kit
- JetPack 5.0.2 R35.1.0 (ubuntu20.04)
- Installing ROS2 Humble by running `./scripts/docker_build_ros.sh --distro humble --package desktop` (https://github.com/dusty-nv/jetson-containers)

I have also followed the suggsetion above to fix the Livox_sdk build, there is no issue about compling Livox_sdk; however when I `colcon build` the livox_ros2_driver from this repo, below error occurs:

```
/opt/ros/humble/install/include/rclcpp/rclcpp/publisher.hpp:81:47: error: static assertion failed: given message type is not compatible with ROS and cannot be used with a Publisher
   81 |     rclcpp::is_ros_compatible_type<MessageT>::value,
```
```
/workspace/ROS2/ws_livox/src/livox_ros2_driver/livox_ros2_driver/lddc.cpp:323:29: error: no matching function for call to ‘rclcpp::Publisher<pcl::PointCloud<pcl::PointXYZI> >::publish(livox_ros::PointCloud&)’
  323 |     publisher->publish(cloud);
```
```
/usr/include/c++/9/type_traits:2378:11: error: no type named ‘type’ in ‘struct std::enable_if<false, void>’
 2378 |     using enable_if_t = typename enable_if<_Cond, _Tp>::type;
```
```
make[2]: *** [CMakeFiles/livox_ros2_driver.dir/build.make:160: CMakeFiles/livox_ros2_driver.dir/livox_ros2_driver/lddc.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/livox_ros2_driver.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< livox_ros2_driver [31.4s, exited with code 2]

Summary: 2 packages finished [34.8s]
  1 package failed: livox_ros2_driver
  1 package had stderr output: livox_ros2_driver
root@ubuntu:/workspace/ROS2/ws_livox# 
```

Any solution for this issue?

Marcus-D-Forte, 2022-09-30:
Peharps you could take a look into

https://github.com/Livox-SDK/livox_ros2_driver/pull/20