Livox-SDK/livox_ros2_driver #23: ROS2 Humble build error

TZECHIN6, 2022-09-23:
### Same error here, below is the environment setup and error reproduce procedures:

Environment:
- Jetson Xavier AGX Developer Kit
- JetPack 5.0.2 R35.1.0 (ubuntu20.04)
- Installing ROS2 Humble by running `./scripts/docker_build_ros.sh --distro humble --package desktop` (https://github.com/dusty-nv/jetson-containers)

I have also followed the suggsetion above to fix the Livox_sdk build, there is no issue about compling Livox_sdk; however when I `colcon build` the livox_ros2_driver from this repo, below error occurs:

```
/opt/ros/humble/install/include/rclcpp/rclcpp/publisher.hpp:81:47: error: static assertion failed: given message type is not compatible with ROS and cannot be used with a Publisher
   81 |     rclcpp::is_ros_compatible_type<MessageT>::value,
```
```
/workspace/ROS2/ws_livox/src/livox_ros2_driver/livox_ros2_driver/lddc.cpp:323:29: error: no matching function for call to ‘rclcpp::Publisher<pcl::PointCloud<pcl::PointXYZI> >::publish(livox_ros::PointCloud&)’
  323 |     publisher->publish(cloud);
```
```
/usr/include/c++/9/type_traits:2378:11: error: no type named ‘type’ in ‘struct std::enable_if<false, void>’
 2378 |     using enable_if_t = typename enable_if<_Cond, _Tp>::type;
```
```
make[2]: *** [CMakeFiles/livox_ros2_driver.dir/build.make:160: CMakeFiles/livox_ros2_driver.dir/livox_ros2_driver/lddc.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/livox_ros2_driver.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< livox_ros2_driver [31.4s, exited with code 2]

Summary: 2 packages finished [34.8s]
  1 package failed: livox_ros2_driver
  1 package had stderr output: livox_ros2_driver
root@ubuntu:/workspace/ROS2/ws_livox# 
```

Any solution for this issue?

_Originally posted by @TZECHIN6 in https://github.com/Livox-SDK/livox_ros2_driver/issues/19#issuecomment-1254469491_

Marcus-D-Forte, 2022-09-30:
Try using this suggestion: https://github.com/Livox-SDK/livox_ros2_driver/pull/20

It worked for me

TZECHIN6, 2022-10-01:
So do you mean it also requires the Livox_SDK to make the livox_ros2_driver works? As I saw there is no comment saying that....

MerdanBay, 2022-10-01:
i had same error, and the suggestion from @Marcus-D-Forte is right.
btw, the [livox_sdk_vendor] pkg will download sdk, so once you compiled it even with error, you will find the sdk in install folder. 
anyway, the easy way before officially merged, you can just use https://github.com/Marcus-D-Forte/livox_ros2_driver version.

Marcus-D-Forte, 2022-10-01:
Yeah, feel free to use my branch. I implemented the fix in both repositories and reference a working livox-sdk branch from livox_ros2_driver.

TZECHIN6, 2022-10-03:
hi @Marcus-D-Forte, I saw your commit on the lddc.cpp file. Why you override the timestamp definition and also using cloud_ros instead of cloud?

TZECHIN6, 2022-10-03:
The colcon build is successful by making below changes:

timesync.h
```
#include <memory>
```
lddc.cpp, line: 319
```
rclcpp::Publisher<sensor_msgs::msg::PointCloud2>::SharedPtr publisher =
      std::dynamic_pointer_cast<rclcpp::Publisher<sensor_msgs::msg::PointCloud2>>
      (GetCurrentPublisher(handle));
  if (kOutputToRos == output_type_) {
    sensor_msgs::msg::PointCloud2 msg;
    pcl::toROSMsg(cloud, msg);
    publisher->publish(msg);
  } else {
```

I am wondering will the commit you made will affect the using experience in ROS2?

Marcus-D-Forte, 2022-10-07:
> hi @Marcus-D-Forte, I saw your commit on the lddc.cpp file. Why you override the timestamp definition and also using cloud_ros instead of cloud?

So this is definetly not required. It helps me synchronize timestamps incoming from external sensors (not from Livox)

Marcus-D-Forte, 2022-10-07:
> The colcon build is successful by making below changes:
>
> timesync.h
>
> ```
> #include <memory>
> ```
>
> lddc.cpp, line: 319
>
> ```
> rclcpp::Publisher<sensor_msgs::msg::PointCloud2>::SharedPtr publisher = [...]

All I can say is that I was able to successfully publish Livox data :). As far as performance goes, it does desenve a comparison investigation.. Maybe https://github.com/ros-perception/perception_pcl/pull/368 has some insight to it.

closed 2023-07-10 by TZECHIN6