Livox-SDK/livox_ros2_driver #26: Is intended to update ROS2 driver to support HAP device?

chechugador, 2023-01-19:
Hi.

I would like to know if there is any information about a possible update to support HAP devices in the ROS2 driver.

Is it a work in progress, a future work that has not started yet, or is there no intention to produce this capability?

Thank you very much in advance.

FPSychotic, 2023-01-19:
I think it is available , please check the site and you will find the new
ROS driver ROS/2 for HAP, mid-360 and the future new lidars.

chechugador, 2023-01-19:
> I think it is available , please check the site and you will find the new ROS driver ROS/2 for HAP, mid-360 and the future new lidars.

Thank you very much!

Yes, as you mentioned, the livox_ros_driver2 includes support for both ROS(1) and ROS2 for the new lidars. I assumed that it was only a ROS1 driver, and that the ROS2 driver was intended to be presented in this repository.

I close the issue.

closed 2023-01-19 by chechugador