Livox-SDK/livox_ros2_driver #28: CustomMsg is invalid.

Griwa88, 2023-02-08:
Platform: ROS2 HUMBLE
I have both livox_ros2_driver and livox_ros_driver2 in my ws_livox

Livox Avia puts out PointCloud2 messages and Custom messages without issue.
PointCloud2 is nicely visualized in RviZ
'ros2 topic echo --once /livox/lidar' returns one custom message without issue under 'livox_interfaces/msg/CustomMsg'

Livox HAP puts out a PointCloud2 message without issue, but I cannot read the Custom message.
PointCloud2 is nicely visualized in RviZ
'ros2 topic echo --once /livox/lidar' returns 'The message type "livox_ros_driver2/msg/CustomMsg" is invalid'

Any suggestions?

FPSychotic, 2023-02-08:
There is a new driver ROS2 for HAP and mid360

Griwa88, 2023-02-08:
I am running latest livox_ros2_driver for Avia, and latest livox_ros_driver2 for HAP. Issue is more towards livox_ros_driver2, but I am not sure if they are related as I am running both.

FPSychotic, 2023-02-08:
You should to use driver2 for HAP with ros1 or ros2, and ros2_driver only
with the Avia, so it looks you are doing right. Good luck in that.

Griwa88, 2023-02-10:
I rebuilt from scratch, seems to have solved the issue.

closed 2023-02-10 by Griwa88