Livox-SDK/livox_ros2_driver #29: Humble ERROR

shenxw0127, 2023-02-13:
Ubuntu22.04
Humble
I have corrected the three files in #20, but the following error occurred
`** WARNING ** io features related to pcap will be disabled
/usr/include/apr-1.0
apr-1
/usr/bin/ld: /usr/local/lib/liblivox_sdk_static.a(device_discovery.cpp.o): relocation R_X86_64_TPOFF32 against symbol `_ZGVZN6spdlog7details2os9thread_idEvE3tid' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: failed to set dynamic section sizes: bad value
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/livox_ros2_driver.dir/build.make:508：liblivox_ros2_driver.so] error 1
gmake[1]: *** [CMakeFiles/Makefile2:139：CMakeFiles/livox_ros2_driver.dir/all] error 2
gmake: *** [Makefile:146：all] error 2
`

shenxw0127, 2023-02-13:
make make CXXFLAGS=-fPIC

closed 2023-02-13 by shenxw0127

li-dahua, 2024-06-03:
Hi, I still have this problem.你怎么解决的？

shenxw0127, 2024-06-04:
> Hi, I still have this problem.你怎么解决的？

IMPORTANT :

If you would like to compile livox_ros2_driver, be sure to delete/rename the "livox_sdk_static.a" file in the "/usr/local/lib/" directory. The livox_ros2_driver would build a shared library of [Livox-SDK](https://github.com/Livox-SDK/Livox-SDK) on its own.

I forgot the IMPORTANT in README. Do not use static library.

ldk-learner, 2024-11-03:
Hi,I renamed the "liblivox_sdk_static.a" file in the "/usr/local/lib/" directory. ,but the following error occurred:
--- stderr: livox_ros2_driver                             
gmake[2]: *** 没有规则可制作目标“/usr/local/lib/liblivox_sdk_static.a”，由“liblivox_ros2_driver.so” 需求。 停止。
gmake[1]: *** [CMakeFiles/Makefile2:139：CMakeFiles/livox_ros2_driver.dir/all] 错误 2
gmake: *** [Makefile:146：all] 错误 2
---
Failed   <<< livox_ros2_driver [0.11s, exited with code 2]

shenxw0127, 2024-11-12:
> ## Hi,I renamed the "liblivox_sdk_static.a" file in the "/usr/local/lib/" directory. ,but the following error occurred:
> --- stderr: livox_ros2_driver
> gmake[2]: *** [...]

感觉你没安sdk啊