Livox-SDK/livox_ros2_driver #39: PointCloud rate issue (CustomFormat)

Eragon123, 2024-03-06:
There seems to be an issue with the Custom PointCloud Format. When I select `xfer_format = 1` then for a `publish_freq` of 50Hz I only get around 27-30Hz and for a `publish_freq` of 10Hz an actual lidar pointcloud rate on the `/livox/lidar` topic of 6.5Hz.
The imu publish frequency always seems to match the expected frequency of 

## What does work:
1. However when I use the ROS1 driver in a different Docker Container on the same system I have no issue with the frame rate and the CPU usage is only around 9% where for the ROS2 driver with CustomPointCloud format and inadequate frequency the CPU usage is around 30%. 
2. When selecting the PointCloud2 format via `xfer_format = 0` the published pointcloud frequency matches the expected lidar publish frequency (e.g. 50Hz) and the CPU usage is also ~5% similar to the ROS1 driver. 

## Possible problem source(s):
- When commenting out the [`FillPointsToCustomMsg`](https://github.com/Livox-SDK/livox_ros2_driver/blob/249be8911010e201f620884ea9d8e7d133a37c0a/livox_ros2_driver/livox_ros2_driver/lddc.cpp#L433) function, the rate for the Custom PointCloud **reaches the expected 50Hz** or 10Hz respectively. The CPU usage in this scenario also goes down to only ~6%. 
I have already tried speeding up the `FillPointsToCustomMsg` function in various ways without any effect on the publish frequency. My understanding is, that with 50packets per published PointCloud and 96 points per packet the loop inside the `FillPointsToCustomMsg` function does 12Million steps per second. However the same function in the [ROS1 Driver](https://github.com/Livox-SDK/livox_ros_driver/blob/3d240d5666129e1a3052e78ee8487a04b08fdda3/livox_ros_driver/livox_ros_driver/lddc.cpp#L356) works just fine!

## Notes:
- Frequencies are observed using `ros2 topic hz`
- The used lidar is a Livox Avia. 
- The observed IMU publish frequency always is `200Hz`
- To build under ROS2 humble the changes from this PR were necessary: [PR](https://github.com/Livox-SDK/livox_ros2_driver/pull/20/files).

Thanks for any help!

curiousityy, 2024-03-18:
@Eragon123, I am facing the same issue with my AVIA. I am also facing an issue with the IMU. My IMU rate is only at 60Hz. May I know what changes you did to get the expected IMU rate?

BloodLemonS, 2024-09-12:
> @Eragon123, I am facing the same issue with my AVIA. I am also facing an issue with the IMU. My IMU rate is only at 60Hz. May I know what changes you did to get the expected IMU rate?

Have you been solve it?

WMAAA, 2025-01-13:
> @Eragon123, I am facing the same issue with my AVIA. I am also facing an issue with the IMU. My IMU rate is only at 60Hz. May I know what changes you did to get the expected IMU rate?

I solved the imu problem, but the lidar custom message still cannot reach 10hz, but strangely, another computer can reach it
I don't know how to solve it