Livox-SDK/livox_ros2_driver #42: Colcon Build Error

ManChrys, 2024-10-06:
[image]

```
khadas@Khadas:~/ws_livox$ colcon build
Starting >>> livox_interfaces
Starting >>> livox_sdk_vendor
Finished <<< livox_interfaces [8.72s]                                     
--- stderr: livox_sdk_vendor                               
In file included from /home/khadas/ws_livox/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp:25:
/home/khadas/ws_livox/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.h:46:8: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
   46 |   std::shared_ptr<std::thread> thread_;
      |        ^~~~~~~~~~
/home/khadas/ws_livox/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.h:30:1: note: ‘std::shared_ptr’ is defined in header ‘<memory>’; did you forget to ‘#include <memory>’?
   29 | #include "noncopyable.h"
  +++ |+#include <memory>
   30 | 
/home/khadas/ws_livox/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp: In member function ‘virtual bool livox::ThreadBase::Start()’:
/home/khadas/ws_livox/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp:34:3: error: ‘thread_’ was not declared in this scope; did you mean ‘pthread_t’?
   34 |   thread_ = std::make_shared<std::thread>(&ThreadBase::ThreadFunc, this);
      |   ^~~~~~~
      |   pthread_t
/home/khadas/ws_livox/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp:34:18: error: ‘make_shared’ is not a member of ‘std’
   34 |   thread_ = std::make_shared<std::thread>(&ThreadBase::ThreadFunc, this);
      |                  ^~~~~~~~~~~
/home/khadas/ws_livox/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp:26:1: note: ‘std::make_shared’ is defined in header ‘<memory>’; did you forget to ‘#include <memory>’?
   25 | #include "thread_base.h"
  +++ |+#include <memory>
   26 | #include <thread>
/home/khadas/ws_livox/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp:34:41: error: expected primary-expression before ‘>’ token
   34 |   thread_ = std::make_shared<std::thread>(&ThreadBase::ThreadFunc, this);
      |                                         ^
/home/khadas/ws_livox/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp:34:68: error: left operand of comma operator has no effect [-Werror=unused-value]
   34 | thread_ = std::make_shared<std::thread>(&ThreadBase::ThreadFunc, this);
      |                                                                  ^~~~

/home/khadas/ws_livox/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp: In member function ‘virtual void livox::ThreadBase::Join()’:
/home/khadas/ws_livox/build/livox_sdk_vendor/livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0/sdk_core/src/base/thread_base.cpp:41:5: error: ‘thread_’ was not declared in this scope; did you mean ‘pthread_t’?
   41 |     thread_->join();
      |     ^~~~~~~
      |     pthread_t
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-c++11-long-long’ may have been intended to silence earlier diagnostics
cc1plus: all warnings being treated as errors
gmake[5]: *** [sdk_core/CMakeFiles/livox_sdk_static.dir/build.make:104: sdk_core/CMakeFiles/livox_sdk_static.dir/src/base/thread_base.cpp.o] Error 1
gmake[5]: *** Waiting for unfinished jobs....
gmake[4]: *** [CMakeFiles/Makefile2:242: sdk_core/CMakeFiles/livox_sdk_static.dir/all] Error 2
gmake[3]: *** [Makefile:136: all] Error 2
gmake[2]: *** [CMakeFiles/livox-sdk-v2.3.0.dir/build.make:86: livox-sdk-v2.3.0-prefix/src/livox-sdk-v2.3.0-stamp/livox-sdk-v2.3.0-build] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/livox-sdk-v2.3.0.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< livox_sdk_vendor [13.1s, exited with code 2]

Summary: 1 package finished [14.1s]
  1 package failed: livox_sdk_vendor
  1 package had stderr output: livox_sdk_vendor
  1 package not processed
```

YHanG505, 2024-11-15:
Just add memorry as what your terminal said.

SAJIB3489, 2024-11-21:
You can check this repository which is compatible with ROS2 Humble and also as a Dockerfile for amd64 and arm64.
**Link:** https://github.com/SAJIB3489/Livox-ros2-humble.git

Livox-louie, 2024-12-17:
maybe you can try using the latest master bransh @ManChrys

closed 2024-12-17 by Livox-louie